CMB27/ModbusRTUSlave #6: Answer delay config parameter

pr4rt0r, 2022-06-21:
Could you add a configuration parameter - response delay? The first byte of the response is lost periodicallly.

CMB27, 2022-06-25:
Okay, feature implemented. Documentation can be found in the [wiki](https://github.com/CMB27/ModbusRTUSlave/wiki/ModbusRTUSlave()). I have also added an [example program](https://github.com/CMB27/ModbusRTUSlave/tree/main/examples/Coil_RS-485) with RS-485, which does include the response delay feature.  
I have not actually tested the feature yet. Please let me know how it works out. I will leave the issue open until you let me know that the feature works, or I have tested it myself.  
Out of curiosity, what hardware are you having an Arduino talk to? It seems to be a little cantankerous.

pr4rt0r, 2022-06-28:
Thank you very much!!! I can't test right now, unfortunately. It was an arduino nano clone on built-in serial (pin1, 2), 232-485 "arduino" module, moxa 1150 and PC.

pr4rt0r, 2022-07-25:
Tested successfully

closed 2022-07-25 by pr4rt0r