CMB27/ModbusRTUSlave #7: Driver control

Fluxanode, 2022-07-19:
What pins are used for transmit enable and receive enable?  
Also what are the INO pins for TX RX?

CMB27, 2022-07-20:
On a half duplex RS-485 transceiver, the transmitter/driver is enabled when its enable pin is high. The receiver is enabled when its enable pin is low. This library is always either transmitting or receiving, so these pins on the transceiver can be connected together. The pin on the Arduino used to control them is specified by you in the library constructor: [`ModbusRTUSlave()`](https://github.com/CMB27/ModbusRTUSlave/wiki/ModbusRTUSlave()).

On the Arduino UNO, if you are using HardwareSerial, RX is pin 0, and TX is pin 1, as can be found on the [Arduino UNO pinout sheet](https://content.arduino.cc/assets/Pinout-UNOrev3_latest.pdf). If you are using SoftwareSerial, you specify the pins you want to use in its constructor: [`SoftwareSerial()`](https://docs.arduino.cc/learn/built-in-libraries/software-serial).

closed 2022-07-20 by CMB27

CMB27, 2022-07-23:
Yes, any Stream object can be used. I am not familiar with the Teensy, but it is likely that its Serial1 is a Stream object and can be used.

In future, please use the Discussion tab for this type of question. The Issues tab is primarily for bug reporting. Also please read over the pages in the wiki tab, there is a lot of detail in there on how to use this library.