CMB27/ModbusRTUSlave #8: Parity isn't working

Max89D, 2022-09-27:
Hello,

Nice lib you've made, I really like it.
I had some issues trying to communicate in 8E1. I constantly had CRC errors on my debug Modbus RTU console.
I've figure out that the config was in fact not 8E1 but 8N1 (as by default). I've had a look at the code and couldn't find where the parity check is done or written. Is it done somewhere else ? I only had a look in the begin method.
Could you resolve this ?

I'm sorry I couldn't get directly into your code and try resolve.

Regards,
Maxim

CMB27, 2022-10-01:
Parity is handled by the [Stream](https://www.arduino.cc/reference/en/language/functions/communication/stream/) object handed to the library in the [constructor](https://github.com/CMB27/ModbusRTUSlave/wiki/ModbusRTUSlave()). The library itself does not check parity. The examples use [SoftwareSerial](https://docs.arduino.cc/learn/built-in-libraries/software-serial) as the stream object, which only supports 8N1. However, if you use a [HardwareSerial (Serial)](https://www.arduino.cc/reference/en/language/functions/communication/serial/) object, 8E1 is supported.

The library needs to know the configuration being used in order to properly handle timing when receiving messages.

Max89D, 2022-10-03:
Thank you very much, it"s very interesting.
I'll try with Serial.

Max89D, 2022-10-04:
I confirm it works with the Serial library but you have to set it in the begin method as following:
`Serial.begin(baud, SERIAL_8E1);`

CMB27, 2022-10-06:
That is how it works.

closed 2022-10-06 by CMB27