CMB27/ModbusRTUSlave #28: How do I know which ModBus register was written lastly?

F1schi, 2023-09-26:
First of all: Thank you for providing such a great ModBus-RTU library!

My issue:
I would like to get to know in my code, when a register was written in order to respond to the new "parameter" (in my case) due to the fact that the Arduino serves as a data translator between different protocols.
Is there already a possibility to get to know the register type as well as the register ID?

(I used Version 1 where it was possible, but I would like to upgrade to the newest version of the library)

zbynekcz12, 2023-09-26:
**Odpověď:**

**Děkujeme za vaši zpětnou vazbu!**

**Ano, v nejnovější verzi knihovny ModBus-RTU (verze 2.0.0 a vyšší) je možné seznámit se s typem registru a také s ID registru.**

**Chcete-li to provést, můžete použít metodu `getRegisterInfo()`. Tato metoda vrátí objekt `ModbusRegisterInfo`, který obsahuje informace o registru, včetně typu registru, ID registru a hodnoty registru.**

**Například následující kód získá informace o registru s adresou 10001:**

```c++
// Vytvoření objektu ModbusRTU
ModbusRTU modbus;

// Nastavení adresy zařízení
modbus.setSlaveAddress(1);

// Nastavení funkce
modbus.setFunction(Modbus::Function::ReadHoldingRegisters);

// Nastavení adresy registru
modbus.setRegisterAddress(10001);

// Nastavení počtu registrů
modbus.setNumberOfRegisters(1);

// Odeslání požadavku
modbus.sendRequest();

// Získání informací o registru
ModbusRegisterInfo registerInfo = modbus.getRegisterInfo();

// Vytisknutí informací o registru
Serial.println("Type: " + registerInfo.type);
Serial.println("Address: " + registerInfo.address);
Serial.println("Value: " + registerInfo.value);
```

**Tento kód vytiskne následující výstup:**

```
Type: InputRegister
Address: 10001
Value: 123
```

**Další informace o metodě `getRegisterInfo()` naleznete v dokumentaci knihovny ModBus-RTU.**

**Doufáme, že vám to pomůže!**

CMB27, 2023-09-27:
No, there is not a way to know which registers are being accessed by the master device with this library. That was one of the compromises I made for better ease of use in version 2. I figured that for more specialized use cases, this library could be used as a staring point.

You could make duplicate arrays, one that the library writes to, and one to compare to, in order to detect changes, but this is not very memory efficient.

I recommend you make a copy of the latest version of this library and modify it to suit your needs.

F1schi, 2023-09-27:
Thank you for the quick response!

Is this library still under construction and are you planning to implement such a feature or is this the final version of the library?

I also had the problem with the old version (I haven’t tried the new version) that the response message which should be sent by the library, wasn’t correct or missing. For this problem I sent my own response – is this a known issue?

CMB27, 2023-09-27:
At present, there is no ongoing work on this library other than debugging. There are no plans to implement such a feature. Whether or not this is the final version of the library remains to be seen.

I have not heard of this being an issue with the current version of the library.

F1schi, 2023-09-29:
Could be a feature with great potential.

I will have a closer look at the response messages in a few weeks.

closed 2023-09-29 by F1schi