CMB27/ModbusRTUSlave #33: 2 byte holding register need fast 100ms

maxmurugan, 2024-02-12:
hi sir

Pico board used  receive data from plc id 01 t0 06 
recive interval -100ms

Modbus library support minimum 300ms duration working

how to read data 100ms speed?

example your  code

#include <ModbusRTUSlave.h>

ModbusRTUSlave modbus(Serial1, 255); // serial port, driver enable pin for rs-485
uint16_t holdingRegisters[2];

union {
  uint16_t bytes[2];
  float x;
} Float_data_1;

void setup() {
  modbus.configureHoldingRegisters(holdingRegisters, 2); // unsigned 16 bit integer array of holding register values, number of holding registers
  modbus.begin(1, 9600);
  Serial.begin(9600);
}

void loop() {
  modbus.poll();

// Split holdingRegisters[0] and holdingRegisters[1]
Float_data_1.bytes[0] = holdingRegisters[1];
Float_data_1.bytes[1] = holdingRegisters[0];

Serial.println(Float_data_1.x); // Print the float value

}

maxmurugan, 2024-02-12:
i set id -1 working 100ms suppose change to id - 06 is speed is slow

CMB27, 2024-02-15:
Are you trying to make the library work faster?

If this is what you are asking, I would suggest increasing your baud rate.
I would set the Modbus baud rate to something like 38400; I've found this to be a good compromise between reliability and speed.

maxmurugan, 2024-02-15:
Yes trying  your library .
I have 6 led board with different id like 01 to 06.
Modbus data send from plc id 1 to 06 one by one every 500ms working fine
change to interval speed 100ms library data showing update slowly

how to receive data every 100ms.

CMB27, 2024-03-01:
Can you please elaborate/clarify on what is going wrong? I am having some trouble understanding exactly what the problem is or even what you are trying to do.

It sounds like you are working with multiple devices and you want communication to be faster. I need more details to be able to help.

closed 2024-03-23 by CMB27