CMB27/ModbusRTUSlave #36: How to catch data have changed event

vanhcm20, 2024-03-23:
Hi @CMB27 
First, Im new in ModbusRTU, frogive me if this is stupid question.
How I can catch when have data changed, i dont want read all data continously

CMB27, 2024-03-23:
@vanhcm20 
Not a stupid question,

The 2.x.x versions of this library do not have a way of directly detecting if data (coils, holding registers, etc.) have changed.
The way you would do this, is to compare your data array(s) from before and after running `poll()`.

This was one of the choices I made in the development of this library going from 1.x.x to 2.x.x.
In 1.x.x you would need to hand functions to the library for reading and writing data, rather than arrays.
This made the library a lot more flexible, but it was also much more cumbersome to setup.

vanhcm20, 2024-03-27:
Thanks for your feedback, i think this will be good function in future , but now, it's good enough to use
Thank you

closed 2024-03-27 by vanhcm20