CMB27/ModbusRTUSlave #39: How to can send 32 bit float

vanhcm20, 2024-04-25:
Thank for your useful library
But how i can send and read 32 bit float?

CMB27, 2024-04-27:
It involves pointers.
A line like this, `float *floatRegisters = (float*)holdingRegisters;` after `uint16_t holdingRegisters[2];` should work.
`floatRegisters` can be treated like an array, just keep in mind that it will have half as many float values than the number of holding registers (so in my example here, 1 float).

CMB27, 2024-04-27:
Here are some examples.

### Master/Client Sending Float
``` C++
#if (defined __AVR_ATmega328P__ || defined __AVR_ATmega168__ || defined __AVR_ATmega1280__ || defined __AVR_ATmega2560__)
  #include <SoftwareSerial.h>
#endif

#include <ModbusRTUMaster.h>

const byte dePin = 13;

#if (defined __AVR_ATmega328P__ || defined __AVR_ATmega168__ || defined __AVR_ATmega1280__ || defined __AVR_ATmega2560__)
  const byte rxPin = 10;
  const byte txPin = 11;
  SoftwareSerial mySerial(rxPin, txPin);
  ModbusRTUMaster modbus(mySerial, dePin);
#elif defined ESP32
  ModbusRTUMaster modbus(Serial0, dePin);
#else
  ModbusRTUMaster modbus(Serial1, dePin);
#endif

uint16_t holdingRegisters[2];
float *floatRegisters = (float*)holdingRegisters;

void setup() {
  modbus.begin(38400);
  Serial.begin(115200);
  Serial.setTimeout(10);
  while(!Serial);
}

void loop() {
  if (Serial.available()) {
    String string = Serial.readStringUntil('\n');
    string.trim();
    floatRegisters[0] = string.toFloat();
    Serial.print("Sending: ");
    Serial.println(floatRegisters[0], 5);
    debug(modbus.writeMultipleHoldingRegisters(1, 0, holdingRegisters, 2));
  }
}

bool debug(bool modbusRequest) {
  if (modbusRequest == true) {
    Serial.println("Success");
  }
  else {
    Serial.print("Failure");
    if (modbus.getTimeoutFlag() == true) {
      Serial.print(": Timeout");
      modbus.clearTimeoutFlag();
    }
    else if (modbus.getExceptionResponse() != 0) {
      Serial.print(": Exception Response ");
      Serial.print(modbus.getExceptionResponse());
      switch (modbus.getExceptionResponse()) {
        case 1:
          Serial.print(" (Illegal Function)");
          break;
        case 2:
          Serial.print(" (Illegal Data Address)");
          break;
        case 3:
          Serial.print(" (Illegal Data Value)");
          break;
        case 4:
          Serial.print(" (Server Device Failure)");
          break;
        default:
          Serial.print(" (Uncommon Exception Response)");
          break;
      }
      modbus.clearExceptionResponse();
    }
    Serial.println();
  }
  Serial.flush();
  return modbusRequest;
}
```

### Slave/Server Receiving Float
``` C++
#if (defined __AVR_ATmega328P__ || defined __AVR_ATmega168__ || defined __AVR_ATmega1280__ || defined __AVR_ATmega2560__)
  #include <SoftwareSerial.h>
#endif

#include <ModbusRTUSlave.h>

const byte dePin = 13;

#if (defined __AVR_ATmega328P__ || defined __AVR_ATmega168__ || defined __AVR_ATmega1280__ || defined __AVR_ATmega2560__)
  const byte rxPin = 10;
  const byte txPin = 11;
  SoftwareSerial mySerial(rxPin, txPin);
  ModbusRTUSlave modbus(mySerial, dePin);
#elif defined ESP32
  ModbusRTUSlave modbus(Serial0, dePin);
#else
  ModbusRTUSlave modbus(Serial1, dePin);
#endif

uint16_t holdingRegisters[2];
float *floatRegisters = (float*)holdingRegisters;

void setup() {
  modbus.configureHoldingRegisters(holdingRegisters, 2); // unsigned 16 bit integer array of holding register values, number of holding registers
  modbus.begin(1, 38400);
  Serial.begin(115200);
  while(!Serial);
}

void loop() {
  float oldFloatValue = floatRegisters[0];
  modbus.poll();
  if (floatRegisters[0] != oldFloatValue) {
    Serial.println(floatRegisters[0], 5);
  }
}
```

vanhcm20, 2024-05-03:
Thank for your reply
Sorry for my late, i done its already but forget to update it

I also use pointer and copy float to uint16_t array
Should keep in mind its use 2 pos in array

That how i send 32 bit float
```
#include <ModbusRTUSlave.h>

ModbusRTUSlave modbus(Serial1);

const int address = 28;
uint16_t holdingRegisters[address];

void setup() {

  Serial.begin(9600);
  modbus.configureHoldingRegisters(holdingRegisters, address); // unsigned 16 bit integer array of holding register values, number of holding registers

  modbus.begin(1, 9600);

  float f = 0.4561;

  memcpy(&holdingRegisters[22], &f, sizeof f);

  holdingRegisters[24] = 10;
}

void loop() {  
  
  modbus.poll();
}
```

closed 2024-05-03 by vanhcm20