CMB27/ModbusRTUSlave #42: Arduino Leonardo and the Serial USB

Antoine2211, 2024-07-15:
Hello, @CMB27

Is it possible to use the library with an Arduino Leonardo and the Serial USB.

regards,

CMB27, 2024-07-15:
It should work, but I haven't tested it.
Don't set a dePin, when you try: use `ModbusRTUSlave modbus(Serial);` instead.

errno, 2024-07-16:
Be aware of Arduino rebooting when built-in USBtoSerial is used.
I stumbled upon this effect just now. I use pymodbus on my Raspberry as master and Arduino Nano connected via USB. Each time I open /dev/ttyUSB0 Arduino restarts. This is required for programming via USB to work and is not related to communication.

Antoine2211, 2024-07-17:
ok it works fine with the Arduino Leonardo <-> PC
Beware on the PC side to open the Serial Port with the option Dtr (DtrEnable at true in C#) if not the Serial.write doesn't work on the Arduino Leonardo side.

closed 2024-07-17 by Antoine2211