CMB27/ModbusRTUSlave #48: Holding registers declaration

Titi95830, 2024-10-10:
Hello
In my Modbus slave project I need to declare the following Holdingregisters addresses : 0x151E, 0x0006, 0x2014,0x101E to respond to Modbus Master requests.

Can you tell me how I should proceed? The following test I made didn't work:

uint16_t holdingRegisters[4] = {0x151E,0x0006,0x2014,0x101E};

and in Loop():
  holdingRegisters[0x151E] = 1,5;
  holdingRegisters[0x0006] = 1500;
  holdingRegisters[0x2014,0x101E] = 1;
  holdingRegisters[0x2014,0x101E] = 30;

Thank you

Translated with DeepL.com (free version)

CMB27, 2024-10-10:
This library only supports continuous holding registers starting with 0.
What you have done with `uint16_t holdingRegisters[4] = {0x151E,0x0006,0x2014,0x101E};` is create four holding registers with addresses 0, 1, 2, and 3, where `holdingRegister[0] == 0x151E`, `holdingRegister[1] == 0x0006`, `holdingRegister[2] == 0x2014`, and `holdingRegister[3] == 0x101E`.

To get the addresses you are looking for you would need to change this to `holdingRegisters[0x2015];`.
This would create 8213 (0x2015) holding registers. This is horribly memory inefficient, considering you are only using four holdingRegisters, but it should work provided you have enough RAM.

I recommend seeing if you can get your master device to request holding registers 0x0000, 0x0001, 0x0002, and 0x0003, rather than 0x151E, 0x0006, 0x2014, and 0x101E.

Titi95830, 2024-10-11:
Hi,
Thanks you very much for these informations.

closed 2024-10-23 by CMB27