CMB27/ModbusRTUSlave #49: Counter value storage

peperounez, 2024-10-19:
Hello,
I included in your ModbusRTUSlaveExample example a counter triggered by 2 switches I would like its value to be stored in a holdingRegisters/inputRegisters. Can we write the value as you did for reading the potentiometers? I don't see it in holdingRegisters[0]/inputRegisters[0] if I use Modbus Poll. Any idea how to do it, because I can see the bit change if I use a coil (coils[0] = !digitalRead(buttonPins[0]))?
int cpt;
void loop()
If....
increment/decrement cpt
end of if

holdingRegisters[0] = cpt; // Storing the counter value
inputRegisters[0] = cpt; // Storing the counter value
end of program

Thanks for the help and the easy to use library.

CMB27, 2024-10-19:
Can you post your code here?

peperounez, 2024-10-22:
[image]
everything works, I have two arduino uno and I simulated the operation of a rotary encoder in the first and I included in the second your program with in addition the reading of 0 and 1 incrementing a counter. The value of this counter is then stored in the registers. Thank you for this easy-to-use library which allows an understanding of what Modbus is and how it works. I put a screenshot which shows the operation. Best Regards

closed 2024-10-22 by CMB27