CMB27/ModbusRTUSlave #51: Add method: setUnitID()

eisenyang, 2024-10-31:
Cool & beautiful coding!

I am working on  auto addressing slaves using your lib.
Could you help me add a method named setUnitId()? 

That will be great!
Thanks!

CMB27, 2024-11-02:
You could run `modbus.begin()` again.
Something like the following.

```C++
#include <ModbusRTUSlave.h>

#define MODBUS_BAUD 38400
#define MODBUS_CONFIG SERIAL_8N1

const uint16_t numHoldingRegisters = 16;
uint16_t holdingRegisters[numHoldingRegisters];

// set initial unitId
holdingRegisters[0] = 1;
uint8_t unitId =  holdingRegisters[0];

ModbusRTUSlave modbus(Serial1);

void setup() {
  modbus.configureHoldingRegisters(holdingRegisters, numHoldingRegisters);
  Serial1.begin(MODBUS_BAUD, MODBUS_CONFIG);
  modbus.begin(MODBUS_BAUD, MODBUS_CONFIG);
}

void loop() {
  modbus.poll();
  if (holdingRegisters[0] != unitId) {
    unitId = holdingRegisters[0];
    modbus.begin(unitId, MODBUS_BAUD, MODBUS_CONFIG);
  }
}

```

eisenyang, 2024-11-03:
Yes, but not semantically.

CMB27, 2024-11-03:
The technical capability you are looking for is already there.
Why do you want a `setUnitId()` function separate from `begin()`?
Please elaborate.

eisenyang, 2024-11-04:
“You know you are working on clean code when each routine you read turns out to be pretty much what you expected. You can call it beautiful code when the code also makes it look like the language was made for the problem.”

— Ward Cunningham

closed 2024-11-04 by CMB27