CMB27/ModbusRTUSlave #52: Problem with THVD1406 Timeout

Armagedon13, 2024-11-04:
Hi i made a program with a [STM32f103c8t6](https://github.com/siyouluo/STM32-Blue-Pill/blob/master/PDF/The-Generic-STM32F103-Pinout-Diagram.pdf)(bluepill) that have 1 master and 1 slave (in differents serials) the master is the MAX485 and works as expected, the [THDV1406](https://www.ti.com/lit/ds/symlink/thvd1406.pdf?ts=1730729588281&ref_url=https%253A%252F%252Fwww.google.com%252F) as slave but i can't make it work, i change the pins DE and RE, all the time it throw timeout error. i test it with master and slave and nothing.

[image]

This is the program it reads all the 2 holding register from slaves from the Master max485 and send it the slave THDV1406
```cpp
#include <HardwareSerial.h>
#include <ModbusRTUMaster.h>
#include <ModbusRTUSlave.h>

// Master Configuration
#define MODBUS_SERIAL_MASTER Serial1
const int8_t dePin_Master = PB12;
ModbusRTUMaster modbus_master(MODBUS_SERIAL_MASTER, dePin_Master);
#define MODBUS_BAUD_MASTER 38400
#define MODBUS_CONFIG_MASTER SERIAL_8N1

// Slave Configuration
HardwareSerial Serial3(PB11, PB10);
#define MODBUS_SERIAL_SLAVE Serial3
const int8_t dePin_slave = PB13;
const int8_t rePin_slave = PB14;
ModbusRTUSlave modbus_slave(MODBUS_SERIAL_SLAVE, dePin_slave, rePin_slave);
#define MODBUS_BAUD_SLAVE 38400
#define MODBUS_CONFIG_SLAVE SERIAL_8N1
#define MODBUS_SLAVE_ID 1  // ID for SCADA communication 

// Modbus Slave Configuration
const uint8_t NUM_SLAVES = 3;  // Number of Modbus slaves to read from, get the size of list later
const uint8_t REGISTERS_PER_SLAVE = 2;  // Number of holding registers to read from each slave
const uint8_t SLAVE_IDS[] = {2, 3, 4};  // Slave IDs to read from

// Holding Registers Buffer
uint16_t holdingRegisters[NUM_SLAVES * REGISTERS_PER_SLAVE];
const uint16_t numHoldingRegisters = NUM_SLAVES * REGISTERS_PER_SLAVE;

// Error Handling
unsigned long transactionCounter = 0;
unsigned long errorCounter = 0;
const char* errorStrings[] = {
    "success",
    "invalid id",
    "invalid buffer",
    "invalid quantity",
    "response timeout",
    "frame error",
    "crc error",
    "unknown comm error",
    "unexpected id",
    "exception response",
    "unexpected function code",
    "unexpected response length",
    "unexpected byte count",
    "unexpected address",
    "unexpected value",
    "unexpected quantity"
};

// Function to log Modbus transactions
void printLog(uint8_t unitId, uint8_t functionCode, uint16_t startingAddress, uint16_t quantity, uint8_t error) {
    transactionCounter++;
    if (error) errorCounter++;
    
    char string[128];
    sprintf(string, "%ld %ld %02X %02X %04X %04X %s", 
            transactionCounter, errorCounter, unitId, functionCode, 
            startingAddress, quantity, errorStrings[error]);
    Serial.print(string);
    
    if (error == MODBUS_RTU_MASTER_EXCEPTION_RESPONSE) {
        sprintf(string, ": %02X", modbus_master.getExceptionResponse());
        Serial.print(string);
    }
    Serial.println();
}

// Read holding registers from all configured slaves
bool readAllSlaveRegisters() {
    uint8_t error;
    uint16_t tempBuffer[REGISTERS_PER_SLAVE];
    
    for (uint8_t i = 0; i < NUM_SLAVES; i++) {
        // Read holding registers from each slave
        error = modbus_master.readHoldingRegisters(
            SLAVE_IDS[i],        // Slave ID
            0,                   // Start address
            tempBuffer,          // Buffer to store registers
            REGISTERS_PER_SLAVE  // Number of registers to read
        );
        
        // Check for errors and log them
        if (error != 0) {
            printLog(SLAVE_IDS[i], 0x03, 0, REGISTERS_PER_SLAVE, error);
            return false;
        }
        
        // Copy registers to consolidated buffer
        memcpy(
            &holdingRegisters[i * REGISTERS_PER_SLAVE], 
            tempBuffer, 
            REGISTERS_PER_SLAVE * sizeof(uint16_t)
        );

        // Log successful read
        printLog(SLAVE_IDS[i], 0x03, 0, REGISTERS_PER_SLAVE, 0);
    }
    
    return true;
}

void setup() {
    // Initialize Serial for debugging
    Serial.begin(115200);
    while(!Serial);
    Serial.println("\n\nModbus RTU Multi-Slave Bridge");
    
    // Initialize Master Serial and Modbus
    MODBUS_SERIAL_MASTER.begin(MODBUS_BAUD_MASTER, MODBUS_CONFIG_MASTER);
    modbus_master.begin(MODBUS_BAUD_MASTER);
    //modbus_master.setTimeout(300);
    
    // Initialize Slave Serial and Modbus
    MODBUS_SERIAL_SLAVE.begin(MODBUS_BAUD_SLAVE, MODBUS_CONFIG_SLAVE);
    modbus_slave.begin(MODBUS_SLAVE_ID, MODBUS_BAUD_SLAVE);
    
    // Initialize holding registers to 0
    memset(holdingRegisters, 0, sizeof(holdingRegisters));

    // Configure holding registers for slave
    modbus_slave.configureHoldingRegisters(holdingRegisters, numHoldingRegisters);
}

void loop() {
    // Read holding registers from all slaves
    readAllSlaveRegisters();
    
    // Process incoming Modbus Slave requests
    modbus_slave.poll();
    
    // Small delay to prevent overwhelming the bus
    delay(100);
}
```

CMB27, 2024-11-06:
Let's make sure I am understanding your issue here.

From what I gather, you are making a sort of gateway device.

This device acts as a Modbus slave/server on one RS-485 bus (lets call this "bus1").
It also acts as a master/client on another RS-485 bus (lets call this "bus2).

The bluepill is connected to bus 1 using a THDV1406 RS-485 transceiver.
It is connected to bus2 using a MAX485 board.

```mermaid
flowchart
master <--> bus1 <--> THDV1406 <--> bluepill <--> MAX485 <--> bus2
bus2 <--> slave2
bus2 <--> slave3
bus2 <--> slave4
```

When setup in this configuration, the master device times out.

Is this correct?

Armagedon13, 2024-11-06:
Yes it's correct

CMB27, 2024-11-06:
I took a look at the datasheet for the THVD1406.
It's an unusual RS-485 transceiver. It doesn't have a driver enable (DE) pin. It instead has a shutdown (SHDN) pin.

Judging by the picture you posted, it looks like the THVD1406 board you are using has the SHDN pin labeled as DE.

In order for the THVD1406 to function at all (sending or receiving) the SHDN pin needs to be pulled high.
According to the datasheet, this pin has a weak internal pullup resistor, so simply disconnecting it may be all you need to do.

Armagedon13, 2024-11-06:
yes! it works. I was making that configuration RE and SHDN high to auto direction! thanks, to summarice i wasn't find the module online and i was searching for 2 days the documentation. it no so fast, i think the max485 is better it doesn't have auto direction but is faster. thanks for the help i was thinking is a library problem but no. sorry for the inconvenience.

closed 2024-11-06 by Armagedon13