CMB27/ModbusRTUSlave #56: Not able to write to holding register or coil

ambalakkandy, 2024-12-23:
I am using the code to convert arduino Nano to modbus RTU slave which communicates with PC via QModmaster software. I am using LInk-150 RS485 to Ethernet Gateway module to connect Arduino to PC. {Arduino TTL output is converted to RS485 signal via a TTL-RS485 onverter. The setup works fine when i read holding registers and coils but write function never works. What can be the issue? Qmodmaster throws an error "Illegal function" when tried to Write to holding register or coil.
`/*
  ModbusRTUSlaveExample

  This example demonstrates how to setup and use the ModbusRTUSlave library (https://github.com/CMB27/ModbusRTUSlave).
  See README.md (https://github.com/CMB27/ModbusRTUMaster/blob/main/examples/ModbusRTUSlaveExample/README.md) for hardware setup details.
  
  Created: 2023-07-22
  By: C. M. Bulliner
  Last Modified: 2024-10-26
  By: C. M. Bulliner
  
*/

#include <ModbusRTUSlave.h>
#define MODBUS_SERIAL Serial

ModbusRTUSlave modbus(MODBUS_SERIAL, 8); //dePin No

bool coils[2];
bool discreteInputs[2];
uint16_t holdingRegisters[3];
uint16_t inputRegisters[2];
uint16_t out;

void setup() {

  //Serial.begin(9600);
//Two numbers of each 
  modbus.configureCoils(coils, 2);
  modbus.configureDiscreteInputs(discreteInputs, 2);
  modbus.configureHoldingRegisters(holdingRegisters, 3);
  modbus.configureInputRegisters(inputRegisters, 2);

  MODBUS_SERIAL.begin(9600, SERIAL_8N1);
  modbus.begin(1, 9600, SERIAL_8N1);
}

void loop() {
holdingRegisters[0] = rand();
holdingRegisters[1] = rand();
out= holdingRegisters[2];
coils[0]=0;
coils[1]=1;
//Serial.println(out);
  modbus.poll();
  //delay(500);
}
`

CMB27, 2024-12-24:
I don't know why QModmaster is throwing an error. Is it displaying any data that could be helpful when debugging?

The Arduino program isn't setup to have the Arduino itself show if any data has been written. It is possible that the write is succeeding on the Arduino. Try something like the following and see if you can control the LED via coil 0.

``` C++
#include <ModbusRTUSlave.h>

ModbusRTUSlave modbus(Serial, 8);
// ModbusRTUSlave modbus(Serial); // If you don't need the DE pin, you can use this line instead.

bool coils[1];

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  modbus.configureCoils(coils, 1);
  Serial.begin(9600); // The config defaults to SERIAL_8N1, so it doesn't need to be set
  modbus.begin(1, 9600); // same here
}

void loop() {
  modbus.poll();
  digitalWrite(LED_BUILTIN, coils[0]);
}
```

ambalakkandy, 2024-12-24:
Thank you for the comment. I had ordered a different gateway module to test the communication again.
I want to add one more observation that that Writing to register /coil is working with following setup
1. Modbus RTU over USB
2.  Modbus RTU over RS485 (i.e. via RS485 to USB converter)

Whether this indicates Modbus configuration for Arduino is correct? 
I will update once i test the setup with new Gateway unit.

ambalakkandy, 2024-12-25:
Update.
Write register function succeeded after changing the Ethernet Gateway module ! So the problem was somewhere in the Configuration setting /hardware of the first Gateway module I used.
Thank you very much for the support.

closed 2025-01-04 by CMB27