CMB27/ModbusRTUSlave #57: Problem implementing on STM32

tnowak, 2025-01-06:
Hi,
I have a hard time implementing this library on a STM32 dev board (Black Pill STM32F411CE).
I'm also using this small PCB with MAX3485 chip: https://sklep.msalamon.pl/wp-content/uploads/2024/12/sklep_msalamon_MOD-00116_18.jpg
Its':
- RX pin is connected to PA9 on Black Pill that seems to be Serial1 TX Pin
- TX pin is connected to PA10 on Black Pill that seems to be Serial1 RX Pin
- EN (?) is most probably RE & DE Pin combined annd is connected to PA4 pin on Black Pill

#define DE_RE_PIN PA4
ModbusRTUSlave modbus_slave(Serial1, DE_RE_PIN);

Serial1.begin(19200, SERIAL_8E1);
modbus_slave.begin(1, 19200, SERIAL_8E1);
[... then I decare some holding registers and do modbus_slave.poll() in the loop.

This very same code worked fine on Arduino Nano Every (but with different RS485 Transceiver - MAX485).
Now I'm trying to use it on STM32 and no Modbus communication is seen on A & B.
What am I missing here?

CMB27, 2025-01-08:
It could be the the RS-485 board. Can you share a picture of the other side of the board or post the transceiver chip model number?

Issue #52 might also give some good insight.

tnowak, 2025-01-08:
@CMB27: here's the board: https://sklep.msalamon.pl/produkt/konwerter-ttl-na-rs485-max3485/
They seems to bee commonly used. The chip is MAX3485. I've already confirmed that EN pin is connected to RE & DE chip pins.

tnowak, 2025-01-08:
This particular transceiver I've used turned out to be broken. Thankfully I've ordered couple of these and on the second one i tried everything works fine. Sorry for bothering you!

closed 2025-01-08 by tnowak