CMB27/ModbusRTUSlave #59: MegaTinyCore problem.

zarss, 2025-04-10:
Hi. I am not sure how to find what is causing my problem.
Sketch compiles for Arduino Nano, no errors. Using MegaTinyCore for Attiny 3224, I am getting strange error: " ... \AppData\Local\Temp\ccO3a2Ka.ltrans0.ltrans.o: In function `_poll_dre':
<artificial>:(.text+0x6d0): undefined reference to `_poll_dre_done' "
If I comment out ModbusTRUSlave library, code compiles. But I cannot anything like _poll_dre_done anywhere in the library. Or anywhere at all.

code: https://pastebin.com/YQVQmdCT
MegaTinyCore: https://github.com/SpenceKonde/megaTinyCore

Thank you, Andris.

CMB27, 2025-04-11:
I just copied your code here.
``` C++
#include <SoftwareSerial.h>
#include <ModbusRTUSlave.h>
SoftwareSerial rs(8, 9); //tx pin, rx pin
ModbusRTUSlave modbus(rs, 10);
 
#include <OneWire.h>
#include <DallasTemperature.h>
#define ONE_WIRE_BUS 1
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);
float tempC = 0.0;
uint16_t inputRegisters[2];
uint32_t task_a;
uint32_t period_a = 5000;
 
void setup() {
 
  modbus.configureInputRegisters(inputRegisters, 2);     // unsigned 16 bit integer array of input register values, number of input registers
  modbus.begin(22, 4800);
  sensors.begin();
}
 
void loop() {
 
  if (millis() - task_a >= period_a) {
    task_a = millis();
    sensors.requestTemperatures();
    tempC = sensors.getTempCByIndex(0);
  }
  inputRegisters[0] = tempC * 10;
  modbus.poll();
}
```

CMB27, 2025-04-11:
`_poll_dre` and `_poll_dre_done` are in the [UART.cpp](https://github.com/SpenceKonde/megaTinyCore/blob/4d0d75660ccfa72de79c9c4f15a8cd17c9f0ed16/megaavr/cores/megatinycore/UART.cpp) file of megaTinyCore.

zarss, 2025-04-11:
Thank you very much, sorted!

closed 2025-04-11 by zarss